OpenBuilds/OpenBuilds-CAM #13: gcode parameter window size

swarfer, 2018-09-19:
Hiya
On my Elitebook the gcode parameter dialog is too short even though the buttons are usable (if you can see them against the background)

The Elitebook is running Ubuntu and its only job is to run the Ox.   I normally generate Gcode on another machine and then transfer though the network, but it woudl be very handy o be able to use the online cam directly at the machine for simple jobs.
[image]

openbuilds-engineer, 2018-09-19:
[image]

openbuilds-engineer, 2018-09-19:
Added a Scrollbar - let me know how it goes (:

closed 2018-09-19 by openbuilds-engineer

swarfer, 2018-09-20:
thanks, will check tonight.
This is an increasing problem with wide screens, they are just not very high in pixel count, usually around 800.

commit fd8a3123 (2018-09-19): Added scrollbar to Operation Modal to fix #13