OpenBuilds/OpenBuilds-CAM #22: MacOS Specifics:  Option vs Control for Multiple Select

openbuilds-engineer, 2018-11-07:
MacOS and its nonstandard keyboard...
As reported by @cojarbi: "fyi in Osx to click select multiple items usually its the option/alt not the control."

openbuilds-engineer, 2018-11-26:
@cojarbi  can you help me check what the browser things the Option key is? 

Open browser, and open Devtools:
Paste 

```
document.addEventListener('keydown', (event) => {
  const keyName = event.key;
  const altKey = event.altKey;
  const ctrlKey = event.ctrlKey
  alert('keydown event\n\n' + 'key: ' + keyName + "  /  Ctrl: " + ctrlKey + "  / altKey: " + altKey );
});
```

Click in the browserwindow and hit that Option key

[image]

Awaiting screenshot

cojarbi, 2018-11-27:
Here you go

openbuilds-engineer, 2018-11-27:
(: nothing uploaded (:

cojarbi, 2018-11-27:
Sended a scrfeenshot

openbuilds-engineer, 2018-11-28:
The Git doesnt understand emails (:   come over to https://github.com/OpenBuilds/cam/issues/22 or PM me you know where I am at

cojarbi, 2018-11-28:
My bad, thought it was an email lol

[image]

openbuilds-engineer, 2018-11-28:
Is Option used instead of Ctrl for things like Undo (Ctrl+Z) and selectAll (Ctrl+A) too?  Or just with clicking?

openbuilds-engineer, 2018-11-28:
Option+click (same as Ctrl+click) is live in five, let me know if it works like it should

closed 2018-11-28 by openbuilds-engineer

cojarbi, 2018-12-14:
Delete Key
[image]

commit f1233531 (2018-11-28): Add keybinding for Option/Alt+Click for MacOS to fix #22