OpenBuilds/OpenBuilds-CAM #43: Unable to open gerbers exprted from easyeda

scaytrase, 2019-03-05:
I'm trying to open my easyeda project to the cam. selecting gerbers gives me

```
main.js:92 New FileOpen Event:
main.js:93 Event
main.js:78 TypeError: Cannot read property 'length' of undefined

(https://cam.openbuilds.com/lib/gerber/gerberparser.js line 8)
gerberparser.js:8 Uncaught TypeError: Cannot read property 'length' of undefined
    at parseGerber (gerberparser.js:8)
    at FileReader.r.onload (main.js:243)
```

also excellon files could not be selected, since no *.DRL format is supported

Gerber archive attached

[Gerber_separate-spindle-board_20190228201511.zip](https://github.com/OpenBuilds/cam/files/2932791/Gerber_separate-spindle-board_20190228201511.zip)

openbuilds-engineer, 2019-03-05:
Only tested with Eagle. 

Checkout https://www.youtube.com/watch?v=0pjSeT0OACI 

Will check your files at some point though, leaving open for now

openbuilds-engineer, 2019-03-05:
Post some screenshots and easyeda files too

scaytrase, 2019-03-05:
it's pretty simple so I expected it to work, but...

[separate-spindle-board.zip](https://github.com/OpenBuilds/cam/files/2933569/separate-spindle-board.zip)

[image]

openbuilds-engineer, 2019-04-22:
Unfortunately at this time I'll close this issue:

1. EasyEDA being a product for LCSC might not spit out correct Gerber X274 format files, it may be custom to their internal workflow so you can order boards from them
2. EasyEDA user base is very limited.  Eagle is huge, Kicad too. 
3. The file fails to parse in https://github.com/OpenBuilds/cam/tree/master/lib/gerber - I simply cannot afford the time to debug why due to the limited user count. Sorry (:

closed 2019-04-22 by openbuilds-engineer

scaytrase, 2019-04-22:
Well, currently I use EasyEDA with FlatCAM + Candle without any issue, it's used not only for ordering boards from JLCPCB

openbuilds-engineer, 2019-04-22:
Cool, glad you got a solution.  As mentioned though, that was a "may be" ^ 
The rest still stands. 

Thanks