OpenBuilds/OpenBuilds-CAM #54: How to make it work offline

Geekoid85, 2019-08-11:
I tried to clone the project and run the index.html that seems to work fine at first. But I can't generate the gcode after I setup an operation. It loads indefinitely.
[image]

The same svg file work fine in the online version.
[image]
Also the text feature does not work offline.

I hope that an offline version can be a future option for this software, it's really easy and fluid. I can test more features as the team fix those bugs, I have some basic knowledge of git python and arduino and access to a grbl cnc machine.

openbuilds-engineer, 2019-08-13:
run it in a local webserver. Chrome doesnt execute scripts from file:// loads

openbuilds-engineer, 2019-08-13:
Quickest: nodejs + http-server https://www.npmjs.com/package/http-server 
Easier https://www.apachefriends.org/index.html (no need for anything but apache, but easy to install)

closed 2019-08-13 by openbuilds-engineer