OpenBuilds/OpenBuilds-CAM #56: Optimize path when planning order of operations.

hydroflame, 2019-09-11:
If I load an image into OpenBuilds-CAM and then ask to generate G-Code it will often generate a non-optimal path (the toolhead will travel more than necessary). It should instead attempt to order the operations in such a way that minimizes job time. Off top my head I know this is a valid optimization to attempt for laser operations but maybe not for drilling or other tool type. Might need some planning.

I made a fashion pattern making software and have already implemented heuristic solutions this problem. (It's the traveling salesman I know). I'd like to try to contribute to OpenBuilds-CAM if public contributions are welcomed.

openbuilds-engineer, 2019-09-17:
Duplicate: https://github.com/OpenBuilds/OpenBuilds-CAM/issues/9

openbuilds-engineer, 2019-09-17:
(: pull requests welcome:  https://github.com/OpenBuilds/OpenBuilds-CAM/blob/master/js/advanced-cam-gcode.js

But yeah, it needs some thinking so you dont change the order of concentrics inside pocket milling, or lines inside a raster etc.   We generate each entitiy on its own, then concatenate inside advanced-cam-gcode.js so kinda best place to look at that

closed 2019-09-17 by openbuilds-engineer