OpenBuilds/OpenBuilds-CAM #58: Center Drill, wrong plunge feedrate used in gcode output

openbuilds-engineer, 2019-10-24:
camFeedrate: "400"
camPlungerate: "5"

[workspace-2019-10-24.zip](https://github.com/OpenBuilds/OpenBuilds-CAM/files/3769309/workspace-2019-10-24.zip)

User says "What I set was a feed rate of 5, that is my confusion here, why is doing it at 400 wheare that nuamber came from"

```
; GCODE Generated by cam.openbuilds.com on 2019-10-21
G21 ; mm-mode
G54; Work Coordinates
G21; mm-mode
G90; Absolute Positioning
M3 S1000; Spindle On

; Operation 0: Drill: Continuous (Centered)
; Tool Diameter: 3.00

G0 Z10
G0 F1000 X14.9674 Y26.9679

G0 Z0
G1 F5 Z0.0000
G1 F400 X14.9674 Y26.9679 Z-6.0000 S1000
G1 F400 X14.9674 Y26.9679 Z0.0000 S1000
; retracting back to z-safe
G0 Z10

M5 S0; Spindle Off
```

https://openbuilds.com/threads/openbuilds-cam-software.13122/page-4#post-88400 relates

naikymen, 2020-11-29:
I have the same issue, and dirty-patched it with a regex in a text-editor:

Pattern: `G1 F1000 (.* Z-5.*)`

Replacement: `G1 F50 \1`

The problem seems to be that (for drilling), the **actual** plunge is done at `cutSpeed` and not at `plungeSpeed` (see the `generateGcode` function at `js/advanced-cam-gcode.js`).

I just wrote "actual plunge" because the **intended** plunge is futile: there is no movement in the Z direction

    G0 Z0
    G1 F20 Z0.0000

Only after those lines, the actual plunge uses the feedrate and not the plungerate:

    G1 F50 X64.6191 Y52.9875 Z-5.0000 S1000
    G1 F50 X64.6191 Y52.9875 Z0.0000 S1000

naikymen, 2020-11-29:
The easyest workaround would be to set `cutSpeed=plungeSpeed` somewhere, only for drilling operations.

naikymen, 2020-11-29:
Untill it is fixed, I managed by:

1. Making a fake milling operation, and setting the feedrate equal to the plungerate.
2. Deleting that operation, and creating the drill operation.

Ta-da!

; Starting : Closed?:true

```
G0 Z10
G0 F1000 X64.6191 Y52.9875

G0 Z0
G1 F20 Z0.0000
G1 F20 X64.6191 Y52.9875 Z-5.0000 S1000
G1 F20 X64.6191 Y52.9875 Z0.0000 S1000
```

closed 2024-04-18 by petervanderwalt