OpenBuilds/OpenBuilds-CAM #61: Pocket tool problem

psyhlo, 2020-02-14:
I tried to use pocket tool for simple figure rectangle
it skips one side of the rectangle in older versions this was working. Last time i use it and it works was in September 2019
[https://prnt.sc/r2a1p9](https://prnt.sc/r2a1p9)

Moemilstead, 2020-02-15:
I am having the same issue.

JinShil, 2020-03-22:
I'm not sure if this is the same issue, but I found that with my pocker operations the gcode generated looks like this:

`; Starting inflatedGroup: Closed?:trueG1 F200 Z-1.0000`

I think there should be a newline between `true` and `G1`.  In the `.gcode` file, I did a search and replace in Notepad++ and it seemed to fix the issue.  The result should be:

```
; Starting inflatedGroup: Closed?:true
G1 F200 Z-1.0000
```
Still, it does appear to be a bug in the CAM software.

petervanderwalt, 2020-03-25:
@JinShil thanks was a little different, but added the Newline as you found it, that was indeed a new bug (initial issue was fixed here: https://github.com/OpenBuilds/OpenBuilds-CAM/commit/6585b615adc044b1dc335da82b757ebbf237f384 I just forgot to close the issue)

commit 05efe070 (2020-03-25): Fix #61 - added Newline after comment
closed 2020-03-25 by petervanderwalt