OpenBuilds/OpenBuilds-CAM #67: Inkscape 1.0 svg resolution

lolivz, 2020-06-26:
Openbuilds-CAM doesn't import svg files created with Inkscape 1.0 at 96 dpi but uses 90 dpi
In Inkscape 1.0 svg file, there is no `` line in XML Header. If I add this line in the svg, it's ok, files are now imported at 96dpi.

petervanderwalt, 2020-07-14:
Post a couple sample files please

lolivz, 2020-07-15:
You'll find 2 identical svg files in this zip : one created with Inkscape 0.92.5, the other with Inkscape 1.0

[inkscape.zip](https://github.com/OpenBuilds/OpenBuilds-CAM/files/4924289/inkscape.zip)

petervanderwalt, 2020-07-15:
Alright, will check it out. 
Found a header we can fingerprint:  ``` inkscape:version="1.0 (4035a4fb49, 2020-05-01)" ```

petervanderwalt, 2020-07-15:
Give it a go, and re-open if you have any issues

closed 2020-07-15 by petervanderwalt

lolivz, 2020-07-15:
It's ok now, many thanks !

commit 07f8ce54 (2020-07-15): Fix #67 Inkscape 1.x DPI