OpenBuilds/OpenBuilds-CAM #70: how to slow stepper motors

williamsenn, 2021-01-28:


swarfer, 2021-01-28:
set the feedrate whne you are setting up the cut parameters
[image]

closed 2021-02-19 by petervanderwalt

williamsenn, 2021-02-20:
Ha yes I learned how to slow the stepper motors down and every job I run works awesome thank y’all for all your help

williamsenn, 2022-02-03:
A-townsmarketplace.com would love your feedback. Post a review to our profile.
https://g.page/r/CXxHel6W6oMqEAo/review

A-townsmarketplace.com

williamsenn, 2024-12-11:
@media all and (-ms-high-contrast: none), (-ms-high-contrast: active) {
  .typography:not(#\#):not(#\#):not(#\#) .markup table.image-wrapper img,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup table.image-wrapper img,
  .typography:not(#\#):not(#\#):not(#\#) .markup table.kindle-wrapper img,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup table.kindle-wrapper img {
    max-width: 550px;
  }
}
@media screen and (max-width: 650px) {
  .typography .markup div.youtube-overlay,
  .typography.editor .markup div.youtube-overlay,
  .typography .markup div.vimeo-overlay,
  .typography.editor .markup div.vimeo-overlay {
    display: none !important;
  }
}
@media screen and (max-width: 370px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup div.tiktok-wrap,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.tiktok-wrap {
    width: calc(95vw - 32px);
    height: calc((95vw - 32px - 2px) / 0.485714);
  }
}
@media screen and (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup div.embedded-publication-wrap .embedded-publication.show-subscribe,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.embedded-publication-wrap .embedded-publication.show-subscribe {
    padding: 24px;
  }
}
@media screen and (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap .subscription-widget.show-subscribe,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap .subscription-widget.show-subscribe,
  .typography:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap-editor .subscription-widget.show-subscribe,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap-editor .subscription-widget.show-subscribe,
  .typography:not(#\#):not(#\#):not(#\#) .markup div.captioned-button-wrap .subscription-widget.show-subscribe,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.captioned-button-wrap .subscription-widget.show-subscribe {
    padding: 0px 24px;
  }
}
@media screen and (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap .subscription-widget.show-subscribe .subscription-widget-subscribe .button,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap .subscription-widget.show-subscribe .subscription-widget-subscribe .button,
  .typography:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap-editor .subscription-widget.show-subscribe .subscription-widget-subscribe .button,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.subscription-widget-wrap-editor .subscription-widget.show-subscribe .subscription-widget-subscribe .button,
  .typography:not(#\#):not(#\#):not(#\#) .markup div.captioned-button-wrap .subscription-widget.show-subscribe .subscription-widget-subscribe .button,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup div.captioned-button-wrap .subscription-widget.show-subscribe .subscription-widget-subscribe .button {
    padding: 10px 12px;
    min-width: 110px;
  }
}
@media (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup .tweet,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup .tweet {
    padding: 12px;
  }
}
@media (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-text,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-text {
    font-size: 14px;
    line-height: 20px;
  }
}
@media (max-width: 650px) {
  .typography:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.two,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.two,
  .typography:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.three,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.three,
  .typography:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.four,
  .typography.editor:not(#\#):not(#\#):not(#\#) .markup .tweet .tweet-photos-container.four {
[... 545 lines cut ...]
    min-width: calc(100% - 20px);
  }
}
@media (max-width: 750px) {
  .poll-embed:not(#\#):not(#\#):not(#\#) .poll-anchor-target .poll-anchor-copy-button {
    left: 8px;
    top: 45px;
  }
}
@media screen and (min-width: 541px) {
  .poll-embed:not(#\#):not(#\#):not(#\#) .poll-wrapper.poll-web .poll-dialog .modal-table .modal-row .modal-content > .container {
    width: 552px;
    padding: 26px 24px;
  }
}
@media screen and (max-width: 650px) {
  .poll-embed:not(#\#):not(#\#):not(#\#) .poll-wrapper.poll-web .poll-dialog .modal-table .modal-row .modal-content > .container {
    padding: 40px 0;
  }
}
@media screen and (max-width: 650px) {
  .poll-embed:not(#\#):not(#\#):not(#\#) .poll-wrapper.poll-web .poll-dialog .modal-row .modal-cell .modal-exit-btn {
    margin-right: -20px;
  }
}You're receiving free posts from DiscreetBoss’s Substack.To get every post and other paid subscriber benefits, upgrade to a paid subscription.Upgrade subscriptionYou can unsubscribe in one click.© 2024 DiscreetBoss548 Market Street PMB 72296, San Francisco, CA 94104 Unsubscribe

williamsenn, 2024-12-11:
View this post on the web at https://discreetboss.substack.com/p/unlock-the-power-of-a-content-calendar

We’ve all been there sitting down to create content and realizing we have no idea what to post next. Whether you're a business owner, marketer, or content creator, staying consistent and strategic with your content can feel like a juggling act. But what if you had a blueprint that made content creation easier and more aligned with your business goals? That’s where a content calendar comes in.
A content calendar isn’t just a tool for scheduling posts. It’s a strategic asset that helps you stay organized, hit your marketing targets, and create content your audience will love. In this article, I’ll walk you through the exact steps to build your own content calendar that’s both simple and effective.
Why a Content Calendar Is a Game Changer
You might be thinking, "Do I really need a content calendar?" The answer is yes, and here’s why:
Consistency is King Regularly posting content is key to building trust with your audience. A content calendar ensures you’re consistently putting out high quality content without scrambling last minute. It helps you plan in advance and stick to a schedule whether that’s weekly blog posts or daily social media updates.
Strategic Planning for Big Wins A calendar allows you to plan content around important dates: product launches, holidays, seasonal trends, or industry events. This means you can tailor your content to what’s most relevant to your audience at any given time.
Focus on Business Goals Without a clear strategy, content creation can feel like spinning your wheels. A content calendar helps you align every post, video, or article with your business objectives whether that’s generating leads, driving traffic, or increasing brand awareness.
How to Build a Killer Content Calendar
Now that you’re sold on why you need a content calendar, let's talk about how to create one. Don't worry it’s easier than you think!
1. Define Your Content Types
Start by deciding what types of content you'll create. Here are some options to consider:
Blog Posts
Social Media Updates (Instagram, Twitter, LinkedIn, etc.)
Video Content (YouTube, TikTok, etc.)
Email Newsletters
Infographics or Visual Content
Understanding your content mix will help you balance different types of posts throughout the month, ensuring you're reaching your audience in a variety of ways.
2. Align Content with Business Goals
Look at your marketing objectives for the next month or quarter. What are you trying to achieve? Lead generation? Brand awareness? Driving traffic to your website?
DiscreetBoss’s Substack is a reader-supported publication. To receive new posts and support my work, consider becoming a free or paid subscriber.
Product Demo Video (Week 1)
Blog Post on Benefits of Product (Week 2)
Customer Testimonial Post (Week 3)
Special Offer Email (Week 4)
Aligning your content with business goals ensures you’re not just posting for the sake of posting you’re creating with a purpose.
3. Map Out Key Dates & Themes
Include any important dates that can guide your content creation. These could be:
Industry Events or Conferences
Holidays (Mother’s Day, Black Friday, etc.)
Product Launches or Anniversaries
Seasonal Themes (Back-to-school, Summer sales, etc.)
For example, if you’re in the fitness industry, you could plan content around New Year’s resolutions, summer fitness goals, or post holiday fitness motivation.
4. Choose Your Tools
Now it’s time to pick the tool that works best for you. Whether you're a visual planner or a list lover, there are tons of tools out there to help you organize your calendar. Here are some options:
Google Sheets or Excel: Simple, customizable, and free. Great for beginners.
Trello: Visual, easy to use boards for planning and organizing content.
Asana: Offers more advanced project management features for teams.
CoSchedule: A premium tool with a content calendar view and social media automation.
Find what works best for you and your team’s workflow!
Content Calendar Example: Put It Into Practice
Let’s bring this to life with an example of how your content might look for the month:
Week 1: Blog post on “How to Overcome [X] Challenge in Your Industry”
Week 2: Instagram post featuring a customer testimonial or success story
Week 3: Email newsletter about an upcoming product launch or promotion
Week 4: YouTube video breaking down a key industry trend
This schedule keeps you on track with varied content types across different platforms. Plus, it's aligned with a broader business goal, like driving traffic to your website or promoting a product.
Why a Content Calendar Will Save Your Sanity (and Your Business)
Having a content calendar isn’t just about staying organized it’s about being strategic. Here’s why it matters:
Stay on Track: No more scrambling for content ideas last minute. You know exactly what’s coming up and when.
Maximize Impact: By planning content around key dates and business objectives, you’re ensuring each piece of content has a purpose.
Boost Consistency: A well planned calendar makes it easy to maintain a consistent content schedule, which is key for building an engaged audience over time.
Action Steps to Get Started
Ready to create your own content calendar? Here’s how to take action today:
Choose Your Tool: Pick the tool that fits your needs Google Sheets, Trello, or something more advanced like Asana or CoSchedule.
[... cut ...]
Unsubscribe https://substack.com/redirect/2/eyJlIjoiaHR0cHM6Ly9kaXNjcmVldGJvc3Muc3Vic3RhY2suY29tL2FjdGlvbi9kaXNhYmxlX2VtYWlsP3Rva2VuPWV5SjFjMlZ5WDJsa0lqb3lPVGd5TmpVd09UZ3NJbkJ2YzNSZmFXUWlPakUxTWprMk1qRTRNaXdpYVdGMElqb3hOek16T1RJek5URTRMQ0psZUhBaU9qRTNOalUwTlRrMU1UZ3NJbWx6Y3lJNkluQjFZaTB6TkRVek56SXhJaXdpYzNWaUlqb2laR2x6WVdKc1pWOWxiV0ZwYkNKOS5nd3NUbHpCUnpBRkhTWTVDYkItdXVHZDRTa2dJMUFmR0JqVFJadk1IQnFZIiwicCI6MTUyOTYyMTgyLCJzIjozNDUzNzIxLCJmIjp0cnVlLCJ1IjoyOTgyNjUwOTgsImlhdCI6MTczMzkyMzUxOCwiZXhwIjoxNzM2NTE1NTE4LCJpc3MiOiJwdWItMCIsInN1YiI6ImxpbmstcmVkaXJlY3QifQ.xcqG0uI_sczjulXHFHBJ-vv6GKuLv9ctSrMDrCBbXCU?

williamsenn, 2024-12-18:
View this post on the web at https://discreetboss.substack.com/p/the-future-of-ai-in-digital-marketing

Hey There, Marketers! 👋
AI isn’t just for sci-fi movies and tech nerds anymore it’s here, and it’s making waves in digital marketing! From creating personalized experiences that make customers feel like VIPs, to predicting what’s going to trend before it even happens, AI is your new marketing BFF.
Let’s explore how AI is going to level up your marketing game, make your life easier, and maybe even make you look like a genius in the process. 🤓✨

Unsubscribe https://substack.com/redirect/2/eyJlIjoiaHR0cHM6Ly9kaXNjcmVldGJvc3Muc3Vic3RhY2suY29tL2FjdGlvbi9kaXNhYmxlX2VtYWlsP3Rva2VuPWV5SjFjMlZ5WDJsa0lqb3lPVGd5TmpVd09UZ3NJbkJ2YzNSZmFXUWlPakUxTXpJNU9UazJOaXdpYVdGMElqb3hOek0wTlRBMk1qZzVMQ0psZUhBaU9qRTNOall3TkRJeU9Ea3NJbWx6Y3lJNkluQjFZaTB6TkRVek56SXhJaXdpYzNWaUlqb2laR2x6WVdKc1pWOWxiV0ZwYkNKOS5feGNiT09ENGN0NS1NNEczYXNtVkx0dnhaU3hQLWpPODBFazlZLVZtRzRVIiwicCI6MTUzMjk5OTY2LCJzIjozNDUzNzIxLCJmIjp0cnVlLCJ1IjoyOTgyNjUwOTgsImlhdCI6MTczNDUwNjI4OSwiZXhwIjoxNzM3MDk4Mjg5LCJpc3MiOiJwdWItMCIsInN1YiI6ImxpbmstcmVkaXJlY3QifQ.0JcoTsDgvYMQ_VhbpOKE6VgCH92Co34QDL0Sy9vRAcY?

williamsenn, 2024-12-18:
View this post on the web at https://discreetboss.substack.com/p/how-ai-helps-your-cro-conversion

In the fast-paced digital world, driving traffic to your website is just the start. The real magic lies in turning that traffic into loyal customers. That’s where Conversion Rate Optimization (CRO) steps in a systematic process of enhancing your website or landing pages to maximize conversions. With the rise of AI, you’ve got a superpower to make this process smoother, smarter, and more impactful. Let’s explore the essentials of CRO and how AI can turbocharge each step.
What is CRO?
At its core, CRO is about transforming clicks into meaningful actions. It’s like upgrading your storefront to ensure that visitors not only enter but also leave with a purchase or at least sign up for your newsletter. The beauty? Even minor tweaks can lead to significant gains in conversions. AI takes this principle to the next level by helping you identify and implement changes faster and more effectively.
Optimizing Landing Pages for Conversions: Step-by-Step with AI

Unsubscribe https://substack.com/redirect/2/eyJlIjoiaHR0cHM6Ly9kaXNjcmVldGJvc3Muc3Vic3RhY2suY29tL2FjdGlvbi9kaXNhYmxlX2VtYWlsP3Rva2VuPWV5SjFjMlZ5WDJsa0lqb3lPVGd5TmpVd09UZ3NJbkJ2YzNSZmFXUWlPakUxTXpNd01ETXpOaXdpYVdGMElqb3hOek0wTlRBMk9UUTVMQ0psZUhBaU9qRTNOall3TkRJNU5Ea3NJbWx6Y3lJNkluQjFZaTB6TkRVek56SXhJaXdpYzNWaUlqb2laR2x6WVdKc1pWOWxiV0ZwYkNKOS44aHRwRW9tODRXUUZUMWVwdmxXSVpiUm9KQURBaXp6SmJMcktPalloNE9JIiwicCI6MTUzMzAwMzM2LCJzIjozNDUzNzIxLCJmIjp0cnVlLCJ1IjoyOTgyNjUwOTgsImlhdCI6MTczNDUwNjk0OSwiZXhwIjoxNzM3MDk4OTQ5LCJpc3MiOiJwdWItMCIsInN1YiI6ImxpbmstcmVkaXJlY3QifQ.D-mN6cnkCF2PoeH-lW_7zYfVy74UIwljZOfLpgZY0Wo?

williamsenn, 2024-12-18:
View this post on the web at https://discreetboss.substack.com/p/a-mindful-approach-to-2024s-top-digital

Have you ever felt like your SEO strategies aren’t quite landing? Picture this: you’re about to launch a major SEO campaign, but before you dive in, you pause. You take a deep breath, clear your mind, and center yourself. That’s mindfulness at work. It’s more than just a stress-relief tool—it can sharpen your focus, elevate your strategy, and give your SEO efforts the edge they need. In 2024, mindfulness could be the game-changer you didn’t know your SEO campaign was missing.
1. Meditation for Marketers: Unlocking Focus & Clarity
When most people think of mindfulness, they picture calm and relaxation. But for marketers, it’s a powerful tool that sharpens mental clarity, enhances resilience, and improves focus—exactly what you need for effective SEO. Studies show that just a few minutes of meditation each day can increase your cognitive function and boost your ability to tackle complex SEO tasks with precision.

Unsubscribe https://substack.com/redirect/2/eyJlIjoiaHR0cHM6Ly9kaXNjcmVldGJvc3Muc3Vic3RhY2suY29tL2FjdGlvbi9kaXNhYmxlX2VtYWlsP3Rva2VuPWV5SjFjMlZ5WDJsa0lqb3lPVGd5TmpVd09UZ3NJbkJ2YzNSZmFXUWlPakUxTXpNd01URTJOaXdpYVdGMElqb3hOek0wTlRBNU1EWTJMQ0psZUhBaU9qRTNOall3TkRVd05qWXNJbWx6Y3lJNkluQjFZaTB6TkRVek56SXhJaXdpYzNWaUlqb2laR2x6WVdKc1pWOWxiV0ZwYkNKOS5lUXVRRGJzVlNRUjhqbzJoWXlxR2pYOElheWwweGpDaUk3OW14b21fQzRzIiwicCI6MTUzMzAxMTY2LCJzIjozNDUzNzIxLCJmIjp0cnVlLCJ1IjoyOTgyNjUwOTgsImlhdCI6MTczNDUwOTA2NiwiZXhwIjoxNzM3MTAxMDY2LCJpc3MiOiJwdWItMCIsInN1YiI6ImxpbmstcmVkaXJlY3QifQ.J7xq6soXWiRGy3u4bkZ_ifW1Xc9uCTlD5boPfRGr9HQ?