OpenBuilds/OpenBuilds-CAM #74: MacOS mouse support

jperrin72, 2021-07-02:
Hi,

I just moved from Windows to MacOS and found out the Mac Mouse is not fully supported. 
When I try to move a shape, the interface both move it and orbit the viewport.

Do you see a workaround or a fix for this ?

Thanks

Jerome

jperrin72, 2021-07-02:
I've managed to reproduce the issue on both Safari and Chrome.
platform is MacOS 11.4 / M1 processor.

jperrin72, 2021-09-12:
anyone care ?

noahwilliamsson, 2021-12-03:
It seems some Mac specific changes were introduced in https://github.com/OpenBuilds/OpenBuilds-CAM/commit/bf7eeb80b892c5e311bbe5e6af1e139a7f6ec284 (Aug, 2018).  Not sure why (touch-pad related?) but perhaps @petervanderwalt can explain?

You can try restoring the functionality by opening the browser's Developer console and pasting the following code into the javascript console.

```
controls.mouseButtons = {
      ORBIT: THREE.MOUSE.MIDDLE,
      ZOOM: false,
      PAN: THREE.MOUSE.RIGHT
    };
```

That should largely undo the changes introduced in the previously mentioned commit, at least if you have an external, non-Apple mouse and run Safari (tested v15.1).  I didn't have much success in Firefox (v94) and the performance was awful.

jperrin72, 2021-12-03:
Thanks Noah, I just tried the workaround and this fix the issue! Do you think this could be appended into the settings so that I don't have to do this each time I use the app ?

petervanderwalt, 2021-12-06:
> Not sure why (touch-pad related?) but perhaps @petervanderwalt can explain?
Long time ago, can't remember exactly, but I do know the fix was solid for every other Mac user since then - no other complaints.

zirkeltraining, 2022-04-28:
can confirm, the problem happens with BOTH Mouse AND Touchpad on a mac M1 MacOS 12.3.1. in Chrome.
And the fix that @noahwilliamsson posted above does solve it for both. 
This should really be merged. Moving objects is impossible otherwise.

jperrin72, 2022-04-28:
> can confirm, the problem happens with BOTH Mouse AND Touchpad on a mac M1 MacOS 12.3.1. in Chrome. And the fix that @noahwilliamsson posted above does solve it for both. This should really be merged. Moving objects is impossible otherwise.

100% agree - this limitation impact all MacOS users.

closed 2024-04-09 by petervanderwalt