OpenBuilds/OpenBuilds-CAM #76: Importing SVG from Inkscape 0.91 has incorrect DPI?

matth, 2021-10-20:
## Problem

Using `Inkscape 0.91 r13725 (Ubuntu 16.04)` the tool imports the SVG at 96dpi, this means a simple SVG with a 50x50 rect gets scaled to a 46.88 x 46.88 mm rect.

If I change [this line](https://github.com/OpenBuilds/OpenBuilds-CAM/blob/07f8ce54bbf0e764572ff3c42d97fccfe28d31e6/js/svgparser.js#L68) to be `>=` the file will be imported at 90dpi and be scaled correctly.

I am pretty interested in scripting the process of SVG -> G-code myself, could you help me understand why the DPI scaling happens or is needed? Is it specific to the OpenBuilds-CAM tool? I can specify in SVG a fixed document width and height in MM or inches, then map the path points to that view the `viewbox` attribute, I don't see why the DPI scaling is required.

Thanks for an awesome tool btw!

## Steps to reproduce

Inkscape version: Inkscape 0.91 r13725 (Ubuntu 16.04)

1. New document
2. File > Document Properties
   - Default Units: mm
3. Draw rect w=50, h=50
4. Remove stroke
5. Path -> Object to Path
6. File -> Document Properties -> Resize page to drawing
7. Save
8. Import into tool and see incorrect sizing

## Example file from Inkscape 0.91

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<!-- Created with Inkscape (http://www.inkscape.org/) -->

<svg
   xmlns:dc="http://purl.org/dc/elements/1.1/"
   xmlns:cc="http://creativecommons.org/ns#"
   xmlns:rdf="http://www.w3.org/1999/02/22-rdf-syntax-ns#"
   xmlns:svg="http://www.w3.org/2000/svg"
   xmlns="http://www.w3.org/2000/svg"
   xmlns:sodipodi="http://sodipodi.sourceforge.net/DTD/sodipodi-0.dtd"
   xmlns:inkscape="http://www.inkscape.org/namespaces/inkscape"
   width="50.100002mm"
   height="50.100002mm"
   viewBox="0 0 177.51969 177.51969"
   id="svg2"
   version="1.1"
   inkscape:version="0.91 r13725"
   sodipodi:docname="inkscape_0.91.svg">
  <defs
     id="defs4" />
  <sodipodi:namedview
     id="base"
     pagecolor="#ffffff"
     bordercolor="#666666"
     borderopacity="1.0"
     inkscape:pageopacity="0.0"
     inkscape:pageshadow="2"
     inkscape:zoom="0.35"
     inkscape:cx="-200.53712"
     inkscape:cy="-214.08604"
     inkscape:document-units="mm"
     inkscape:current-layer="layer1"
     showgrid="false"
     fit-margin-top="0"
     fit-margin-left="0"
     fit-margin-right="0"
     fit-margin-bottom="0"
     inkscape:window-width="1983"
     inkscape:window-height="1128"
     inkscape:window-x="65"
     inkscape:window-y="24"
     inkscape:window-maximized="1" />
  <metadata
     id="metadata7">
    <rdf:RDF>
      <cc:Work
         rdf:about="">
        <dc:format>image/svg+xml</dc:format>
        <dc:type
           rdf:resource="http://purl.org/dc/dcmitype/StillImage" />
        <dc:title></dc:title>
      </cc:Work>
    </rdf:RDF>
  </metadata>
  <g
     inkscape:label="Layer 1"
     inkscape:groupmode="layer"
     id="layer1"
     transform="translate(-96.965695,-140.75647)">
    <path
       style="fill:#000000;fill-opacity:1;stroke:#000000;stroke-width:0.35433072;stroke-miterlimit:4;stroke-dasharray:none;stroke-opacity:1"
       d="m 97.14286,140.93364 177.16536,0 0,177.16536 -177.16536,0 z"
       id="rect4000"
       inkscape:connector-curvature="0" />
  </g>
</svg>
```

matth, 2021-10-20:
FWIW, I think the `96dpi` change came in for version `0.92`:

https://wiki.inkscape.org/wiki/Release_notes/0.92

petervanderwalt, 2022-01-07:
> Using `Inkscape 0.91`

Upgrade to latest Inkscape, 1.0 has been out for 2 years now

closed 2022-01-07 by petervanderwalt