OpenBuilds/OpenBuilds-CAM #79: Issue with Inkscape 1.1.x

hiss-tone, 2022-02-27:
Hi,
related issue has been discussed here a few times but seems that OpenBuild-CAM is parsing the SVG document in wrong DPI. For some reason values are parsed (or compared?) incorrectly. In 1.0.2 it works fine but not with version 1.1.2. If I just edit the header inside the file and set the version to 1.0.2 then it scales correctly.

In version 1.1.2 I have this line:
inkscape:version="1.1.2 (b8e25be833, 2022-02-05)"
and in 1.0.2 like this:
inkscape:version="1.0.2-2 (e86c870879, 2021-01-15)"

If I manually edit the 1.1.2 header to this:
inkscape:version="1.0.2 (b8e25be833, 2022-02-05)"
it works fine!

Something probably goes wrong around these lines of code:
https://github.com/OpenBuilds/OpenBuilds-CAM/blob/f86c7cc39a340962fb9adc8e7ba1c7e2dd6639cd/js/svgparser.js#L58
https://github.com/OpenBuilds/OpenBuilds-CAM/blob/f86c7cc39a340962fb9adc8e7ba1c7e2dd6639cd/js/svgparser.js#L66
and it gets the wrong resolution...

hiss-tone, 2022-02-27:
Hmm,
took another look using browser console with this result:
----
main.js:2 New FileOpen Event:
main.js:2 Event {isTrusted: true, type: 'change', target: input#file, currentTarget: input#file, eventPhase: 2, …}
svgparser.js:2 File {name: 'test_1.0.2.svg', lastModified: 1646002673191, lastModifiedDate: Mon Feb 28 2022 00:57:53 GMT+0200 (Eastern European Standard Time), webkitRelativePath: '', size: 1938, …}
svgparser.js:11 File: test_1.0.2.svg was created in inkscape version 1.  Setting import resolution to 90dpi
----
main.js:2 New FileOpen Event:
main.js:2 Event {isTrusted: true, type: 'change', target: input#file, currentTarget: input#file, eventPhase: 2, …}
svgparser.js:2 File {name: 'test_1.1.2.svg', lastModified: 1646002659224, lastModifiedDate: Mon Feb 28 2022 00:57:39 GMT+0200 (Eastern European Standard Time), webkitRelativePath: '', size: 1515, …}
svgparser.js:11 File: test_1.1.2.svg was created in inkscape version 1.1.  Setting import resolution to 96dpi
----
Looks like CAM application is not correctly converting the version value and comparison fails. 90dpi results to correct image.

petervanderwalt, 2022-08-23:
https://github.com/OpenBuilds/OpenBuilds-CAM/commit/5ead1e6e9c7d603f24e06ada66dd025a01868b89

closed 2022-08-23 by petervanderwalt

hiss-tone, 2022-08-24:
Super, thanks a lot!

hiss-tone, 2023-02-24:
Looks like this issue is back again. Suddenly the scaling went wrong and it has not been an issue after the above change :(

hiss-tone, 2023-02-25:
Hmm, need to investigate a bit more. With a new test file it renders correctly but the older (saved with new version of Inkscape) is scaled wrongly... I'll get back.

petervanderwalt, 2023-02-27:
> but the older (saved with new version of Inkscape)

Possibly old version has different default DPI than new version. SVGs and DPIs are forever painful, even within the Inkscape ecosystem.  The change in DPI (and originally the nonstandard default) will haunt them for a while