OpenBuilds/OpenBuilds-CAM #87: incorect objects while importing DXF file

petrasl1976, 2023-03-05:
i am getting incorrect objects (arcs bend to opposite side that original) drawn while importing DXF file. I use OnShape to draw and export DXF

[image]
[image]

Could it be that i exporting it incorrectly?

Ger-55, 2023-03-06:
Hi, i have the same problem with circles in Openbuilds CAM.  You can check you dxf file with a Dxf viewer program . I use Adobe Illustrator for design and the dxf is ok but CAM messes up things. The Gcode commands to draw curves are rather complex .

petrasl1976, 2023-03-07:
DXF viewers show properly. I think issue is with the OpenBuilds CAM while displaying it.

Ger-55, 2023-04-05:
As far as I know OpenBuilds CAM converts arc segments into small straight lines, and there things go wrong. Unfortunately I have very little knowledge on JavaScript so checking the source code for this is not for me.

Majormakes, 2023-04-06:
any solution to this problem, still enduring the same issue

petervanderwalt, 2024-04-09:
Just give it clean files per https://docs.openbuilds.com/doku.php?id=docs:software:file-errors

closed 2024-04-09 by petervanderwalt