OpenBuilds/OpenBuilds-CAM #107: problème de programme

micha-georges, 2025-08-15:
cela fait des année que je travaille avec openbuilds pour un graveur numérique, et voici quelque mois qu'il ne répond plus, même pour pour les images que j'ai déjà gravé.
A son allumage il y a un message que je ne comprend pas pour quoi, et avec cela, je n'arrive plus a faire un "Gcode", j'ai voulu le réinstaller, mais il n'y a pas de changement, que dois je faire ?? 

voici le message qui est sur openguilds:
[09:11:17] [ ERROR ] No WebGL Support found on this computer! Disabled 3D Viewer - Sorry!

[09:11:17] [ ERROR ] Your graphics card does not seem to support WebGL: See http://khronos.org/webgl/wiki/Getting_a_WebGL_Implementation to learn more

[09:11:17] [ websocket ] Bidirectional Websocket Interface Started Succesfully

micha-georges, 2025-08-15:
je ne vois vraiment pas comment je doit faire avec "markown" vu que je n'arrive m'aine pas a entrer ou lire ce qui ce passe sur le programme !!!!!!

swarfer, 2025-08-16:
# 🔧 Étapes pour réactiver WebGL dans OpenBuilds

## 1. Vérifier si WebGL est actif dans le navigateur
- Ouvrez **Chrome ou Edge** et tapez : `chrome://gpu`  
- Ouvrez **Firefox** et tapez : `about:support`  
- Regardez la ligne **WebGL**.  
  - Si c’est marqué *désactivé* ou *software only*, la carte graphique n’est pas reconnue correctement.  

## 2. Mettre à jour les pilotes de la carte graphique
- Vérifiez quelle carte graphique vous avez :  
  - Sous **Windows** : appuyez sur `Win + R`, tapez `dxdiag`, puis regardez l’onglet **Affichage**.  
- Téléchargez les pilotes sur le site du fabricant :  
  - [Intel Driver & Support Assistant](https://www.intel.com/content/www/fr/fr/support/intel-driver-support-assistant.html)  
  - [Pilotes GeForce NVIDIA](https://www.nvidia.com/Download/index.aspx?lang=fr)  
  - [Pilotes AMD](https://www.amd.com/fr/support)  
- Installez le pilote et **redémarrez** l’ordinateur.  

## 3. Activer l’accélération matérielle dans le navigateur
- **Chrome/Edge** : Paramètres → Système → *Utiliser l’accélération matérielle si disponible* (activez, puis redémarrez).  
- **Firefox** : Paramètres → Général → Performance → *Utiliser l’accélération graphique matérielle*.  

## 4. Vérifier les mises à jour Windows
- Allez dans **Paramètres → Mise à jour et sécurité → Windows Update** et installez toutes les mises à jour.  

## 5. Tester WebGL
- Allez sur [https://get.webgl.org/](https://get.webgl.org/)  
- Si vous voyez le cube tourner, **WebGL fonctionne**.  
- Redémarrez OpenBuilds CONTROL.  

---

## ⚠️ Si ça ne marche toujours pas
- La carte graphique est peut-être trop ancienne, ou les pilotes ne sont plus supportés.  
- Dans ce cas, OpenBuilds peut encore générer du **G-code**, mais sans l’aperçu 3D.  
- Autre solution : essayer OpenBuilds CONTROL sur un **autre ordinateur plus récent** (même un portable pas cher suffit).

swarfer, 2025-08-16:
sorry, I'm lost now, maybe @petervanderwalt can help you go further.

You can probably use Google or another search engine to find more information on activating WebGL, even an AI like ChatGPT can be used as a search engine to find instructions, as I demonstrated.  Just give it a good problem description and it will probably give you a solution, if there is something you don't understand, just ask it for more detail.

petervanderwalt, 2025-08-16:
Try a different computer with a decent GPU/drivers that. Supports WebGl

micha-georges, 2025-08-17:
l'ordi avec lequel je fait les &quot;gcods&quot; n'est pas le m'aime que 
pour le graveur, j'ai testé sur le 2 ordi et plus rien ne va, alors 
qu'il y a 6mois, j'ai fait pas male de gravure, mais aujourd'hui je 
n'arrive plus a g